beagleboard/Latest-Images #98: Upstream WPANUSB

jadonk, 2022-01-31:
**Is your feature request related to a problem? Please describe.**
BeagleConnect Freedom relies on WPANUSB or BCFSERIAL drivers. It would be great if they were upstream and "just worked" rather than working with an out-of-tree fork.

**Describe the solution you'd like**
I got this feedback from Stefan Schmidt on what needs to be done:
What work need to be done to get this in good shape and upstreamed?
Quite a few tasks, sadly.

Random order:
1. The wpanusb linux driver is missing quite a few driver ops (listen
before talk, set frame retries). I added rough sketches but these are
non functional due to 2.
2. The zephyr firmware side needs to implement these ops as well and
hook it up to the zephyr radio_api
3. A way to read the permanent extended address from the device to use
in Linux is needed
4. The generic aspect of the firmware driver combo is missing
completely. Band, supported channel, power levels, etc is all hard-coded
in the linux driver and not queried from the device firmware. Its not
even giving the page on which a channel is being set when changing it.
5. More driver ops are coming in right now as there is heavy work
towards support management frames as well as active and passive scanning
on the Linux stack. Which means these would need proper handling on teh
firmware side as well.

I would say 1-3 are the simpler ones here and should have a clear
strategy to get fixed.

The whole generic aspect is a lot more complex. We would want this
driver to be able to cover all supported devices on the zephyr side and
communicate the capabilities of the currently used radio to the linux
driver so the linux stack can be setup correctly. We also would want to
set this up in a way to allow for future extensions like new frequency
bands in newer spec versions, and so on.

And none of this is taking a bigger MTU variant into account.

Number 5 also means that we would need someone to actively maintain the
zephyr side of wpanusb and keep up which changes.

**Describe alternatives you've considered**
There's no substitute for upstream.

**Additional context**
This might make a good internship project. Overall, I think we should start doing a better job of putting our backlog here in a public place.

silver2row, 2022-02-02:
Hello Everyone,

I am willing to take time out of my awkward schedule to assist in Open Source activities. I have found, through learning a bit at a time, that Open Source is more than just a philosophy and practice. It is also a way for people to congregate around Linux, programming, and STEM. Not only in education but also in real life practice, Open Source initiatives have helped me post-education to learn about what little was offered while in school on the subjects involving but not limited to STEM in general.

Outside of that plea, I would like to associate myself with the beagleboard.org persons and help while learning new aspects to the ZyphrProject and building in and around Linux.

Seth

P.S. I actually slept on it. I was unaware of what would entail and it frightened me. But, like most procedures, one has to skip on the scared aspect to it all. The only concern I have currently is the BCFSERIAL driver(s). I have been researching ideas on the Project, how to build, and the driver aspect to it would be the most time consuming because I know very little about where in the world this specific driver is located or even how to build it. If there are questions or concerns on your part for me specifically, please let me know. I will answer!

jadonk, 2022-02-04:
Our current fork lives at https://github.com/jadonk/wpanusb/tree/beagleconnect

There's also our serial implementation at https://github.com/jadonk/bcfserial

jadonk, 2022-02-04:
Some mailing list discussion: https://lore.kernel.org/linux-wpan/e75f3fd8-fa0c-7708-e914-f757965920c0@datenfreihafen.org/T/

silver2row, 2022-02-05:
Hello...okay. So, would I need a CC1352r/dev. board for messing w/ the wpan-USB driver and some USB datasheet in particular?

I noticed a Quark SE C1000 dev. board in question but they are a bit tiresome to find for now. I will keep trying.

Seth

P.S. I am sticking w/ it. If you have more time to bother, just send some prerequisites for me. I will research ideas.

jadonk, 2022-02-05:
I can get you a dev board if you'll do this work. Contact me on Slack (find the link for GSoC students) to provide an address.

silver2row, 2022-02-06:
Hello...I would not boast about me but I can try. I will contact you on Slack.

silver2row, 2022-02-13:
Hello Sir,

I am close to doing things. Anyway, here are some errors I have received from trying to install things and make things happen, e.g. in case you or anyone else may wonder about how or why things are transpiring:

```

# From Building the WPAN lib. found at https://github.com/linux-wpan/wpan-tools
# This was on the BBGG w/ git clone and then building...

./configure: line 4898: syntax error near unexpected token `newline'
./configure: line 4898: `LT_INIT('

Also, on an instance of Ubuntu for the Win 11 Machine, I found these items to be evident:

phlic@Morphie:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2047:0aa5 Texas Instruments BeagleConnect
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
phlic@Morphie:~$ dmesg | tail
[ 2873.429957] usb 1-1: New USB device found, idVendor=2047, idProduct=0aa5, bcdDevice= 2.00
[ 2873.429959] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2873.429960] usb 1-1: Product: BeagleConnect
[ 2873.429961] usb 1-1: Manufacturer: BeagleBoard.org
[ 2873.429962] usb 1-1: SerialNumber: 29B59A4621000900
[ 2873.436333] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 2896.577670] init: (144) ERROR: LogException:33: TELEMETRY: (null) No such file or directory @d:\os\src\onecore\VM\linux\mountutil\mountutilcpp.h:19 (MountEnum)

[ 2975.218625] usb 1-1: seqnum max
[ 2998.882359] init: (4288) ERROR: InitCreateProcessUtilityVm:1777: read failed -1 9

```

So, I have not used wpan-next or wpan-tools before and building is a bit different for now. 

Learning at its finest!

Seth

P.S. The python3 script is elusive still to me: ` cc2538-bsl.py /usr/share/beagleconnect/cc1352/wpanusb_beagleconnect.bin /dev/ttyACM0 ` .

I have not found it nor do I know what it looks like so far. So, on the image from your beagleconnect repo. that I found, things almost work out of the box but...

1. I cannot use the `bcfserial` repo. or `wpanusb` repo. either. 
2. I am not completely sure what the click id board is currently. I went to their site online, Mikroe, and could not find the exact replica. Oh and I think it is a "debugger" of sorts. I am not sure even though the side click section(s) are listed and w/ many GNDs.

A. I have some plans.
B. I will report back soon.

RobertCNelson, 2022-02-13:
@silver2row WSL is not a target anyone in the community can 'patch' please post those issues to the WSL github...

Test images are located here:
https://rcn-ee.net/rootfs/debian-mikrobus-armhf/

Regards,

RobertCNelson, 2022-02-14:
@silver2row saw your messages on irc, this is a prototype, please don't ask on irc in the general category..

silver2row, 2022-02-14:
@RobertCNelson ,

Yes sir. No issue.

Seth