beagleboard/Latest-Images #100: Kernel update not finding Seeed-modules

MrBillium, 2022-03-08:
I just installed the latest release eMMC IOT image on a new BB Green ( Seeed studio). This is brand new bare device with nothing installed yet.

 bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img

My very first action was to perform kernel update but I  ran into an issue with updating per instructions:

**cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh**

Pull succeeded and the script ran but gave the following warnings at the end

E: Unable to locate package seeed-modules-4.19.94-ti-r71
E: Couldn't find any package by glob 'seeed-modules-4.19.94-ti-r71'
E: Couldn't find any package by regex 'seeed-modules-4.19.94-ti-r71'

Full Update  output:

debian@beaglebone:/opt/scripts/tools$ sudo ./update_kernel.sh
info: Updating libgnutls30
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Get:3 http://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Hit:4 http://repos.rcn-ee.com/debian buster InRelease
Fetched 65.4 kB in 5s (13.4 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree
Reading state information... Done
libgnutls30 is already the newest version (3.6.7-4+deb10u7).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
info: checking archive
2022-03-08 20:17:34 URL:https://rcn-ee.com/repos/latest/buster-armhf/LATEST-ti [196/196] -> "LATEST-ti" [1]
-----------------------------
Kernel Options:
ABI:1 LTS414 4.14.108-ti-r144
ABI:1 LTS419 4.19.94-ti-r71
ABI:1 LTS54 5.4.106-ti-r38
ABI:1 LTS510 5.10.100-ti-r37
-----------------------------
Kernel version options:
-----------------------------
LTS414: --lts-4_14
LTS419: --lts-4_19
LTS54: --lts-5_4
LTS510: --lts-5_10
-----------------------------
info: you are running: [4.19.94-ti-r71], latest is: [4.19.94-ti-r71] updating...
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://deb.debian.org/debian-security buster/updates InRelease
Hit:4 http://repos.rcn-ee.com/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Checking dpkg...
Checking apt-cache...
debug: reinstalling: [linux-image-4.19.94-ti-r71]
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 28.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf linux-image-4.19.94-ti-r71 armhf 1buster [28.6 MB]
Fetched 28.6 MB in 4s (6,910 kB/s)
(Reading database ... 76212 files and directories currently installed.)
Preparing to unpack .../linux-image-4.19.94-ti-r71_1buster_armhf.deb ...
Unpacking linux-image-4.19.94-ti-r71 (1buster) over (1buster) ...
Setting up linux-image-4.19.94-ti-r71 (1buster) ...
update-initramfs: Generating /boot/initrd.img-4.19.94-ti-r71
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 6,056 B of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf libpruio-modules-4.19.94-ti-r71 armhf 1buster [6,056 B]
Fetched 6,056 B in 0s (36.3 kB/s)
(Reading database ... 76212 files and directories currently installed.)
Preparing to unpack .../libpruio-modules-4.19.94-ti-r71_1buster_armhf.deb ...
Unpacking libpruio-modules-4.19.94-ti-r71 (1buster) over (1buster) ...
Setting up libpruio-modules-4.19.94-ti-r71 (1buster) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 101 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf ti-sgx-ti335x-modules-4.19.94-ti-r71 armhf 1buster [101 kB]
Fetched 101 kB in 0s (310 kB/s)
(Reading database ... 76212 files and directories currently installed.)
Preparing to unpack .../ti-sgx-ti335x-modules-4.19.94-ti-r71_1buster_armhf.deb ...
Unpacking ti-sgx-ti335x-modules-4.19.94-ti-r71 (1buster) over (1buster) ...
Setting up ti-sgx-ti335x-modules-4.19.94-ti-r71 (1buster) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package seeed-modules-4.19.94-ti-r71
E: Couldn't find any package by glob 'seeed-modules-4.19.94-ti-r71'
E: Couldn't find any package by regex 'seeed-modules-4.19.94-ti-r71'
update-initramfs: Generating /boot/initrd.img-4.19.94-ti-r71
debian@beaglebone:/opt/scripts/tools$

Here is my Version script output

git:/opt/scripts/:[a335abcf87d2ef5fd96e7de83cdf3f0ff5a4da2b]
eeprom:[A335BNLTBBG1BBG121010534]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[4.19.94-ti-r71]
nodejs:[v10.24.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-customizations]:[1.20220301.0-0~buster+20220301]
pkg:[bb-usb-gadgets]:[1.20220112.3-0~buster+20220112]
pkg:[bb-wl18xx-firmware]:[1.20211222.2-0~buster+20211222]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
dmesg | grep pru
dmesg | grep pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
debian@beaglebone:/opt/scripts/tools$

RobertCNelson, 2022-03-08:
Hi @MrBillium i'm done with that script.. in your case please use:

```
sudo apt update
sudo apt install bbb.io-kernel-4.19-ti-am335x
```

It'll take care of all kernel updates as they get pushed out..

Regards,

MrBillium, 2022-03-09:
Thanks.

You should  consider updating  instructions at
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian

Bill