beagleboard/Latest-Images #103: no flasher in bone-eMMC-flasher-debian-10.12-console images

rickyketch, 2022-08-04:
The flasher is enabled in uEnv.txt:

```
debian@beaglebone:~$ tail -3 /boot/uEnv.txt
##enable Generic eMMC Flasher:
cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

but nothing happens.  Probably because the scripts are missing:
```
debian@beaglebone:~$ ls /opt/scripts
ls: cannot access 'opt/scripts': No such file or directory
```

I tried all of these console image snapshots:
bone-eMMC-flasher-debian-10.12-console-armhf-2022-06-02-1gb.img.xz
bone-eMMC-flasher-debian-10.12-console-armhf-2022-07-01-1gb.img.xz
bone-eMMC-flasher-debian-10.12-console-armhf-2022-08-01-1gb.img.xz
all the same, no flasher.

Can I just copy /opt/scripts from the IOT snapshot, or is other stuff also missing?

Thank you!
rix

RobertCNelson, 2022-08-04:
Sorry, regenerating the console image...  I've been moving things to a non /opt/scripts/ method for flashing, looks like i didn't finish that in buster.

Should have something uploaded in a hour or two.

Regards,

closed 2022-08-04 by RobertCNelson

RobertCNelson, 2022-08-05:
@rickyketch please verfy with:

https://rcn-ee.net/rootfs/bb.org/testing/2022-08-04/buster-console-armhf/

and i'll update the forum post..

thank you!

rickyketch, 2022-08-06:
I tested bone-eMMC-flasher-debian-10.12-console-armhf-**2022-08-04**-1gb.img.xz.
eMMC flashing worked as expected.
Thank you Robert!