beagleboard/Latest-Images #106: cc2538-bsl.py is broken for BeaglePlay

Ayush1325, 2023-06-05:
**Describe the bug**
The `cc2538-bsl.py` script shipped with [BeaglePlay debian image](https://www.beagleboard.org/distros/beagleplay-debian-11-flasher) does not work for BeaglePlay CC1352 flashing.

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz)**
[am625x-emmc-flasher-debian-11.6-xfce-arm64-2023-02-04-10gb.img.xz](https://files.beagle.cc/file/beagleboard-public-2021/images/am625x-emmc-flasher-debian-11.6-xfce-arm64-2023-02-04-10gb.img.xz)

**Describe your hardware**
[BeaglePlay](https://beagleboard.org/play)

**Describe any Capes Installed**

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
This script is not present.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile a sample ([`samples/hello_world`](https://git.beagleboard.org/beagleconnect/zephyr/zephyr/-/tree/sdk/samples/hello_world)) for target `beagleplay_cc1352`.
2. Flash the `zephyr.bin` file using `cc2538-bsl.py`.

**Expected behavior**
A successful flash:
```sh
❯ python cc2538-bsl.py zephyr.bin /dev/ttyS4
CC2538 BSL script customized for Laughing Coyote CC1352P target!
Setting BOOT and RESET low
Setting RESET high
Setting BOOT high
Opening port /dev/ttyS4, baud 50000
Reading data from zephyr.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC1350 PG2.0 (7x7mm): 704KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x000AFFD8
Primary IEEE Address: 00:12:4B:00:29:B9:98:1A
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 720896 bytes starting at address 0x00000000
Write 208 bytes at 0x000AFF300
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0x8da1b3cc)
```

**Actual behavior**
```sh
❯ cc2538-bsl.py zephyr.bin /dev/ttyS4
Opening port /dev/ttyS4, baud 50000
Reading data from zephyr.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

**Additional context**
The script present [here](https://git.beagleboard.org/beagleconnect/zephyr/zephyr/-/blob/260cc8bb2da3aa38334df045821c0867a5a9fd99/boards/arm/beagle_play_cc1352/cc2538-bsl.py) works fine for BeaglePlay.

RobertCNelson, 2023-06-05:
i see you were on discord, for the next user, you need to load the overlay to disable auto-loading:

```
/overlays/k3-am625-beagleplay-bcfserial-no-firmware.dtbo
```

Regards,

jadonk, 2023-06-06:
I have a new utility at https://git.beagleboard.org/beagleconnect/cc1352-flasher on the from-20230521 branch that is better at supporting multiple devices and has cleaned-up commits. I'm not sure what else I need to do to make it user-friendly and where I can promote it as an alternative upstream to cc2538-bsl.py.

cc @Andrei