beagleboard/Latest-Images #109: GPIO not working on the BeagleBoneAI64 with firmware versions after April 2023

kevinacahalan, 2023-10-05:
**GPIO does not work on the BeagleBoneAI64 with firmware versions after April 2023. It neither works from Linux, or from an R5 core.**

With some no not so thorough testing, I have found that with the images before `bbai64-emmc-flasher-debian-11.6-minimal-arm64-2023-04-06-4gb.img.xz` inclusive, I am still able to control GPIO. With the images after `bbai64-emmc-flasher-debian-11.6-minimal-arm64-2023-04-06-4gb.img.xz` I can not control GPIO.

- Here's a GPIO toggle program for testing GPIO from an R5 https://github.com/FredEckert/bbai64_cortex-r5_example/tree/r5_toggle.
- GPIO test stuff for Linux [gpiod.zip](https://github.com/beagleboard/Latest-Images/files/12822102/gpiod.zip). Make sure to run the C program, some of the scripts did not work for whatever reason.

I initially hit this bug after I updated my BBAI64 and ran the eMMC firmware update scripts.

Take a look at the #beaglebone-ai-64 channel in Discord to see discussion.

jadonk, 2023-10-23:
@RobertCNelson is this a BeagleBoard-DeviceTrees regression? @lorforlinux - Can you check this?

RobertCNelson, 2023-10-23:
@jadonk yeap it's a regression, it's a little fun as it seems to have been caused by the r5 firwmare/u-boot combination..